nopnop2002/esp-idf-ssd1306 #11: Flip Display Orientation

raf1312, 2021-04-23:
Hi,
On my lilygo Cam the Display is mounted upside down. 
Is there any easy solution to flip the image orientation 180°?

Thank you in advance. =)

nopnop2002, 2021-04-24:
Try latest.

[image]

[image]

closed 2021-06-03 by nopnop2002

marchingband, 2024-09-07:
This doesn't work for me with an SPI display.

nopnop2002, 2024-09-07:
@marchingband 
It's work for me.
[image]

marchingband, 2024-09-08:
Turns out I have SH1106, for me it mirrors vertically, so the test is upside down compared to before, but also reversed.
Possibly just because its a different display?

nopnop2002, 2024-09-08:
>Possibly just because its a different display?

I think it's because the hardware implementation is different from the standard SSD1306.

marchingband, 2024-09-08:
interesting, this is a custom PCB from a client.
In any case I got u8g2 to work, so I am good now.
Let me know if there's anything I can do.

nopnop2002, 2024-09-08:
There is a possibility that it will function correctly if you change the Memory Addressing Mode (20h), but I have not tried it.

```
	spi_master_write_command(dev, OLED_CMD_SET_MEMORY_ADDR_MODE);	// 20
	spi_master_write_command(dev, OLED_CMD_SET_PAGE_ADDR_MODE);		// 02
```