nopnop2002/esp-idf-ssd1306 #16: TTGO LORA32 T3_V1.6: I2C reset

vogelfreiheit, 2022-04-25:
Hi!

I've been trying to use your component but unfortunately despite following both your instructions and these:
https://github.com/Xinyuan-LilyGO/TTGO-LoRa-Series

I am getting a reset:

```
␛[0;32mI (596) main.c: INTERFACE is i2c␛[0m
␛[0;32mI (606) main.c: CONFIG_SDA_GPIO=4␛[0m
␛[0;32mI (606) main.c: CONFIG_SCL_GPIO=15␛[0m
␛[0;32mI (616) main.c: CONFIG_RESET_GPIO=16␛[0m
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```
My GPIO config:

```
#define OLED_RST    -1
#define OLED_SDA    21
#define OLED_SCL    22

#define CONFIG_SDA_GPIO OLED_SDA
#define CONFIG_SCL_GPIO OLED_SCL
#define CONFIG_RESET_GPIO OLED_RST
```

Using platformIO, so I have them defined as CFLAGS -D per target board.

nopnop2002, 2022-04-25:
>Using platformIO

Are you useing esp-idf framework?

```
$ cat platformio.ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
```

vogelfreiheit, 2022-04-25:
Yes, in case this helps: I solved it by doing the initialization before everything else except for nvs. Apparently something blocks and the watchdog for the wifi task resets the MCU.

closed 2022-04-25 by vogelfreiheit