nopnop2002/esp-idf-ssd1306 #18: Request: provide API to draw XBM bitmap from monochrome buffer with given height/width

vogelfreiheit, 2022-04-25:
The API should be a functional equivalent to:

https://github.com/olikraus/u8g2/wiki/u8g2reference#drawxbm=

> Description: Draw a [XBM Bitmap](http://en.wikipedia.org/wiki/X_BitMap). Position (x,y) is the upper left corner of the bitmap. XBM contains monochrome, 1-bit bitmaps.
> The current color index is used for drawing (see [setColorIndex](https://github.com/olikraus/u8g2/wiki/userreference#setcolorindex)) pixel values 1. Version 2.15.x of U8g2 introduces a solid and a transparent mode for bitmaps. By default, drawXBM will draw solid bitmaps. This differs from the previous versions: Use setBitmapMode(1) to switch to the previous behavior. The XBMP version of this procedure expects the bitmap to be in PROGMEM area (AVR only). Many tools (including GIMP) can save a bitmap as XBM. A nice step by step instruction is [here (external link)](https://sandhansblog.wordpress.com/2017/04/16/interfacing-displaying-a-custom-graphic-on-an-0-96-i2c-oled/). The result will look like this:

```
#define u8g_logo_width 38
#define u8g_logo_height 24
static unsigned char u8g_logo_bits[] = {
   0xff, 0xff, 0xff, 0xff, 0x3f, 0xff, 0xff, 0xff, 0xff, 0x3f, 0xe0, 0xe0,
...
   0xff, 0x3f, 0xff, 0xff, 0xff, 0xff, 0x3f, 0xff, 0xff, 0xff, 0xff, 0x3f };
```

vogelfreiheit, 2022-04-25:
The abstraction example can be found here:

nopnop2002, 2022-04-25:
>The API should be a functional equivalent to:

I don't thin so.

vogelfreiheit, 2022-04-25:
If you can't or don't want to put the time into writing it, that is fine and entirely up to you, I don't have the time to write it myself right now.

nopnop2002, 2022-04-25:
There are two ways to control the ssd1306.

1. Output in page units.
It works fast with less memory

2. Redraw the entire TFT
8 pages * 128 segment buffer is required.

A second method is needed to draw at Position (x, y).

vogelfreiheit, 2022-04-25:
Trying to find the source of the API, but this example is more accurate than the one I posted before:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/XBM/XBM.ino

A function that is dependent on one single width seems a bit limited to me.

nopnop2002, 2022-04-26:
I found that XBM bitmap format is very usefull.

vogelfreiheit, 2022-04-26:
Excellent! Your repositories are a blessing, is there a way I can send you coffee/beer/bento money?

nopnop2002, 2022-04-26:
>is there a way I can send you coffee/beer/bento money?

No, thanks.

closed 2022-08-09 by nopnop2002