nopnop2002/esp-idf-ssd1306 #19: Hurray for ssd1306_bitmaps()

cvonk, 2022-04-27:
**Thanks for the bitmaps function**. I am using it!

Just ran into something that could use some attention.

In the new function
```c
void ssd1306_bitmaps(SSD1306_t * dev, int xpos, int ypos, uint8_t * bitmap, int width, int height, bool invert)
```
The `xpos`, `ypos` and `height` parameters are in pixels. I suggest either changing the parameter `width` to mean width in pixels, or renaming it so that it is clear that it is pixels/8.

Also, the `ImageDemo` example calls
```c
ssd1306_bitmaps(&dev, xpos, ypos, batman, 4, 12, false);
```
It passes the height as `12`, but the image in the array `batman[]` is 13 pixels heigh.

/Coert

nopnop2002, 2022-04-27:
I accepted your suggestion and changed the unit of the width parameter from bytes to pixels.

However, width only supports multiples of 8.

```
//ssd1306_bitmaps(&dev, xpos, ypos, batman, 4, 12, false);
ssd1306_bitmaps(&dev, xpos, ypos, batman, 32, 13, false);
```

nopnop2002, 2022-05-01:
I found that ssd1306_bitmaps can't be drawn too quickly.
Please try this.

https://github.com/nopnop2002/esp-idf-ssd1306/tree/master/AnimationDemo
https://github.com/nopnop2002/esp-idf-ssd1306/tree/master/MeterDemo

cvonk, 2022-05-05:
Will look, but I couldn't use the bitmaps, because they clear the screen.
I resorted back to displaying a 16x8 bitmap using two character bitmaps.

Cheers,
/c

cvonk, 2022-05-05:
Noticed you run SPI_CLK at 1 MHz.  Surprised you need short wires.  Remember seeing it should go up to 8 or 10 MHz.

nopnop2002, 2022-05-05:
I tried it at a frequency of 2M but it didn't draw correctly in my module.