nopnop2002/esp-idf-ssd1306 #20: I don't know any Init cmd and I have any CMD_STREAM or DATA_STREAM question

WangShuoran, 2022-08-12:
When in your code:
https://github.com/nopnop2002/esp-idf-ssd1306/blob/cfd3ba8fd35009ce9edc890594bde5112678f68f/components/ssd1306/ssd1306_i2c.c#L62

```c
	i2c_master_write_byte(cmd, OLED_CMD_SET_DISPLAY_OFFSET, true);		// D3
	i2c_master_write_byte(cmd, 0x00, true);
	i2c_master_write_byte(cmd, OLED_CONTROL_BYTE_DATA_STREAM, true);	// 40
	//i2c_master_write_byte(cmd, OLED_CMD_SET_SEGMENT_REMAP, true);		// A1
	if (dev->_flip) {
		i2c_master_write_byte(cmd, OLED_CMD_SET_SEGMENT_REMAP_0, true);		// A0
	} else {
		i2c_master_write_byte(cmd, OLED_CMD_SET_SEGMENT_REMAP_1, true);		// A1
	}
	i2c_master_write_byte(cmd, OLED_CMD_SET_COM_SCAN_MODE, true);		// C8
	i2c_master_write_byte(cmd, OLED_CMD_SET_DISPLAY_CLK_DIV, true);		// D5
	i2c_master_write_byte(cmd, 0x80, true);
	i2c_master_write_byte(cmd, OLED_CMD_SET_COM_PIN_MAP, true);			// DA
```
I think when OLED_CONTROL_BYTE_DATA_STREAM those below code will can not happen, because 0x40 let ssd1306 think those below code is data instead of cmd.

Can you give me any answer about this.

When you use ssd1306, your code let any OLED_CONTROL_BYTE_DATA_STREAM and OLED_CONTROL_BYTE_CMD_STREAM in your i2c function begin and end. Those cmd is necessary?

And I have another question, when I see the ssd1306 manual, the D/C# flag begin with Control Byte, I want to know this cmd is legal?
[image]

```c
#define OLED_CONTROL_BYTE_CMD_SINGLE    0x80
#define OLED_CONTROL_BYTE_CMD_STREAM    0x00
#define OLED_CONTROL_BYTE_DATA_SINGLE   0xC0
#define OLED_CONTROL_BYTE_DATA_STREAM   0x40
```

nopnop2002, 2022-08-12:
Thank you for reporting.

>I think when OLED_CONTROL_BYTE_DATA_STREAM those below code will can not happen, because 0x40 let ssd1306 think those below code is data instead of cmd.＞

It was OLED_CMD_SET_DISPLAY_START_LINE instead of OLED_CONTROL_BYTE_DATA_STREAM.
Both are 0x40.

```
    //i2c_master_write_byte(cmd, OLED_CONTROL_BYTE_DATA_STREAM, true);  // 40
    i2c_master_write_byte(cmd, OLED_CMD_SET_DISPLAY_START_LINE, true);  // 40
```

>And I have another question, when I see the ssd1306 manual, the D/C# flag begin with Control Byte, I want to know this cmd is legal?

```
/* Control byte for i2c
Co : bit 8 : Continuation Bit
 * 1 = no-continuation (only one byte to follow)
 * 0 = the controller should expect a stream of bytes.
D/C# : bit 7 : Data/Command Select bit
 * 1 = the next byte or byte stream will be Data.
 * 0 = a Command byte or byte stream will be coming up next.
 Bits 6-0 will be all zeros.
Usage:
0x80 : Single Command byte
0x00 : Command Stream
0xC0 : Single Data byte
0x40 : Data Stream
*/
```

WangShuoran, 2022-08-12:
I know, so can I think your code in my question.
https://github.com/nopnop2002/esp-idf-ssd1306/blob/cfd3ba8fd35009ce9edc890594bde5112678f68f/components/ssd1306/ssd1306_i2c.c#L62
This cmd is useless, because of your answer, the OLED_CONTROL_BYTE_DATA_STREAM have no happen.

Sorry about I didn't say my point in a polite way

WangShuoran, 2022-08-13:
I know, Thank you for your answer  patiently.

closed 2022-08-13 by WangShuoran