nopnop2002/esp-idf-ssd1306 #22: Proposal to Separate SSD1306 Component to a Standalone Repository

Krzyshio, 2023-05-29:
Hi @nopnop2002,

I've been using your SSD1306 Driver for esp-idf and find it very useful. 
However, I noticed that the SSD1306 driver is bundled together with numerous examples within the same repository.

Although these examples provide valuable context, I suggest separating the SSD1306 driver into its own dedicated repository. This would make it easier for developers to integrate the driver into their projects, possibly as a Git submodule, without the necessity of cloning the entire current repository.

Separating the driver into its own repository could streamline its adoption by other developers and make it more accessible.

Please consider this proposal and let me know your thoughts.

Thanks for your time,

nopnop2002, 2023-05-29:
Drivers are only here.

https://github.com/nopnop2002/esp-idf-ssd1306/tree/master/components/ssd1306

Krzyshio, 2023-05-29:
Thanks for your quick response and for the explanation on how to use sparse checkout to isolate the driver.

However, the method you suggested lacks one important feature for developers, that is direct linkage to your original repository.

By using your driver as a standalone repository or as a Git submodule, each project that uses your driver can easily reference back to the original repository, allowing other developers to understand where the driver came from and perhaps contribute to its development.

Furthermore, with a direct link to the original repository, developers can watch for updates, issues, and discussions regarding the driver, keeping them in the loop for any potential changes or improvements. This would not be possible with the current method you've suggested as the link between the original repository and the cloned directory is lost.

Lastly, with a Git submodule or standalone repository, updating to the latest version of the driver is more straightforward. Instead of manually performing the sparse checkout for newer commits, developers can use Git's built-in mechanisms to update submodules, ensuring they have the latest version with less manual intervention.

I believe these benefits justify considering the usage of a standalone repository or Git submodule for your SSD1306 driver. Please let me know your thoughts on this.

nopnop2002, 2023-05-29:
ESP-IDF behaves differently depending on the version.

You can think of V4.X and V5.X as completely different things.

In this case, create a branch in the repository and manage the source for each version.

Fortunately, SSD1306 can be handled by one branch, but we may need to separate branches.

Even if we used a standalone repository for the SSD1306 driver, we still need to branch it.

Krzyshio, 2023-05-29:
Despite these challenges, I still believe that separating the SSD1306 driver into its own repository and using Git submodules could be beneficial. This would not eliminate the need for maintaining separate branches for different ESP-IDF versions, but it could simplify the process of integrating your driver into other projects.

When your driver is used as a Git submodule in a project, developers can easily switch between the branches in your repository to match their ESP-IDF version. This allows them to work with the version of your driver that is compatible with their ESP-IDF version, while also maintaining a direct link to your repository.

Additionally, the use of Git submodules can make the process of updating to the latest version of your driver much simpler. Instead of manually performing the sparse checkout for newer commits, developers can use Git's built-in mechanisms to update the submodule, ensuring they always have the latest compatible version with less manual intervention.

nopnop2002, 2023-12-19:
@digiexchris 

I would like you to show me the specific directory structure so that there is no misunderstanding.

digiexchris, 2023-12-19:
here's a good example https://github.com/eldendiss/DendoStepper

which you can install by either cloning directly into your project's
components directory, or using the idf.py add-dependency tool, or
downloading and unzipping the files. The includes directory is just
specified in the cmakelists file, so you can put it anywhere even the
current directory.

nopnop2002, 2023-12-19:
@digiexchris 

Thank you for your quick response.
I'll consider it.

nopnop2002, 2023-12-28:
I saw this.
https://github.com/eldendiss/DendoStepper

This directory structure is similar to that commonly found in Arduino libraries.

On the other hand, the official esp-idf sample makes extensive use of component directories.
```
./zigbee/esp_zigbee_rcp/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/home/nop/esp-idf/components
./zigbee/esp_zigbee_rcp/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/home/nop/esp-idf/components
./zigbee/esp_zigbee_gateway/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/home/nop/esp-idf/components
./zigbee/esp_zigbee_gateway/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/home/nop/esp-idf/components
./wifi/antenna/components
./ethernet/enc28j60/components
./ethernet/basic/components
./bluetooth/nimble/throughput_app/blecent_throughput/components
./bluetooth/esp_ble_mesh/ble_mesh_coex_test/components
./bluetooth/esp_ble_mesh/aligenie_demo/components
./system/unit_test/components
./system/console/advanced/components
./system/gcov/components
./peripherals/spi_master/hd_eeprom/components
./peripherals/dedicated_gpio/soft_uart/components
./peripherals/dedicated_gpio/soft_i2c/components
./peripherals/dedicated_gpio/soft_spi/components
./peripherals/gpio/generic_gpio/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/home/nop/esp-idf/components
./peripherals/gpio/generic_gpio/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/home/nop/esp-idf/components
./peripherals/gpio/matrix_keyboard/components
./storage/sd_card/sdmmc/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/home/nop/esp-idf/components
./storage/sd_card/sdmmc/build/esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/home/nop/esp-idf/components
./storage/custom_flash_driver/components
./build_system/cmake/plugins/components
./build_system/cmake/import_prebuilt/prebuilt/components
./build_system/cmake/import_lib/components
```

digiexchris, 2023-12-30:
yes and that works fine, as long as the cmakelists.txt file in the root of the repo resolves those paths correctly.

bullestock, 2024-05-09:
It is easy enough to use this project as a submodule in an ESP-IDF project:

```
cd components
mkdir ssd1306
cd ssd1306
git submodule add https://github.com/nopnop2002/esp-idf-ssd1306.git
```
Now, in the top level `CMakeLists.txt`, add

```
set(EXTRA_COMPONENT_DIRS
    ./components/ssd1306/esp-idf-ssd1306/components/ssd1306/)
```

nopnop2002, 2024-09-08:
There are two ways to use this library in your project.

The first is the method described by bullestock.
This method allows you to use this library in all your projects.

Second, you can enable this library only for specific projects.
```
cd $HOME
git clone https://github.com/nopnop2002/esp-idf-ssd1306
cd esp-idf-ssd1306
cp -r components your_project_directiory
```

ermvad, 2025-02-04:
As a solution create idf_component.yml near main.c with this:
```
dependencies:
  ssd1306:
    path: components/ssd1306/
    git: https://github.com/nopnop2002/esp-idf-ssd1306.git
```
When you build a projects esp-idf will automaticly fetch repository to _managed_component_ dir and link with your code

ermvad, 2025-02-05:
@nopnop2002 however i don't know behaviour, when the source code in repo located in other branch (not main/master), and don't know if the esp-idf automaticly update repo when some commit was made

nopnop2002, 2025-02-05:
@ermvad 

We will let you know once we know what happens when the component is updated.

nopnop2002, 2025-02-11:
@ermvad 

When the remote master side was updated, the following message was displayed during build.

```
NOTICE:
Following dependencies have new versions available:
Dependency "nopnop2002/RCSwitch": "6f9fe7f068239f02b0a0de2c4e4d693f9612949d" -> "fd6d8d02ccb6597dd8d82449accd1327c0eccc4a"
Consider running "idf.py update-dependencies" to update your lock file.
NOTICE: Processing 2 dependencies:
NOTICE: [1/2] nopnop2002/RCSwitch (6f9fe7f068239f02b0a0de2c4e4d693f9612949d)
NOTICE: [2/2] idf (5.3.2)
```

nopnop2002, 2025-04-11:
I think ermvad's suggestion is the best one.
```
dependencies:
  ssd1306:
    path: components/ssd1306/
    git: https://github.com/nopnop2002/esp-idf-ssd1306.git
```

I'm closing this issue.

closed 2025-04-11 by nopnop2002