nopnop2002/esp-idf-ssd1306 #29: Will this work with RTC?

hsnprojects, 2024-09-06:
DS3231 in the following repository is using a different strategy to initialize i2c bus.
https://github.com/nopnop2002/esp-idf-ds3231

We can't initialize i2c bus twice, am I right?
Do we need to configure anything differently to use the OLED display along with RTC?

Thank you for these libraries that saved me a ton of time!

nopnop2002, 2024-09-06:
>We can't initialize i2c bus twice, am I right?

It is necessary to initialize each i2c bus.

```
	ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_0, &i2c_config));
	ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0));

	ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_1, &i2c_config));
	ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_1, I2C_MODE_MASTER, 0, 0, 0));
```

hsnprojects, 2024-09-06:
Thanks for taking the time to reply to my message.

Do we still need to include i2cdev from DS3231? It has its own i2c_master_init() function, which conflicts with the one from SSD1306. I noticed that ESP-IDF itself has a different I2C bus initialization function (i2c_new_master_bus(&bus_config, &i2c_bus)).

Can all three of these co-exist?

nopnop2002, 2024-09-06:
>It has its own i2c_master_init() function, which conflicts with the one from SSD1306.

Yes, you are correct.
```
$ cd esp-odf-ssd1306

$ cp -r components/ /home/nop/rtos/esp-idf-ds3231

$ cd /home/nop/rtos/esp-idf-ds3231/components/ssd1306
$ vi ssd1306.h
#change from i2c_master_init to i2c_master_init_ssd1306

$ vi ssd1306_i2c_legacy.c
#change from i2c_master_init to i2c_master_init_ssd1306

$ vi ssd1306_i2c_new.c
#change from i2c_master_init to i2c_master_init_ssd1306

$ cd ../..

$ cd main

$ vi main.c
# add this to end of code
    SSD1306_t dev;

#if CONFIG_I2C_INTERFACE
    ESP_LOGI(TAG, "INTERFACE is i2c");
    ESP_LOGI(TAG, "CONFIG_SDA_GPIO=%d",CONFIG_SDA_GPIO);
    ESP_LOGI(TAG, "CONFIG_SCL_GPIO=%d",CONFIG_SCL_GPIO);
    ESP_LOGI(TAG, "CONFIG_RESET_GPIO=%d",CONFIG_RESET_GPIO);
    i2c_master_init_ssd1306(&dev, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO, CONFIG_RESET_GPIO);
#endif // CONFIG_I2C_INTERFACE

    ssd1306_clear_screen(&dev, false);
    ssd1306_contrast(&dev, 0xff);
    ssd1306_display_text_x3(&dev, 0, "Hello", 5, false);
    vTaskDelay(3000 / portTICK_PERIOD_MS);
```

hsnprojects, 2024-09-07:
Fantastic! That resolved my original issue.

But I am still unable to include protocol_examples_common.h properly. I am very new to the CMake build system.

Following is what I have in my project root folder's CMakeLists.txt:
Could you take a look at this to see if you notice anything glaring?
 
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)
set(EXTRA_COMPONENT_DIRS ./components/ssd1306)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(ds3231)

nopnop2002, 2024-09-07:
This is my CMakeLists.txt
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# This example uses an extra component for common functions such as Wi-Fi and Ethernet connection.
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(ds3231)
```

hsnprojects, 2024-09-07:
Here is the latest error:
E (558) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver

But it looks like we are using ssd1306_i2c_new.c

hsnprojects, 2024-09-07:
By the way, I have another similar board working with these components, but I am forcing it to use the legacy driver, and it works fine with esp-idf v5.3.

nopnop2002, 2024-09-07:
vi components/ssd1306/CMakeLists.txt

```
set(component_srcs "ssd1306.c" "ssd1306_spi.c")

# get IDF version for comparison
set(idf_version "${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}")

#if(idf_version VERSION_GREATER_EQUAL "5.2")
#   list(APPEND component_srcs "ssd1306_i2c_new.c")
#else()
#   list(APPEND component_srcs "ssd1306_i2c_legacy.c")
#endif()
list(APPEND component_srcs "ssd1306_i2c_legacy.c")

idf_component_register(SRCS "${component_srcs}" PRIV_REQUIRES driver INCLUDE_DIRS ".")
```

Please specify the GPIO number of SSD1306 directly.
```
void ssd1306(void *pvParameters)
{
    SSD1306_t dev;

    ESP_LOGI(TAG, "INTERFACE is i2c");
    ESP_LOGI(TAG, "CONFIG_SDA_GPIO=%d",CONFIG_SDA_GPIO);
    ESP_LOGI(TAG, "CONFIG_SCL_GPIO=%d",CONFIG_SCL_GPIO);
    ESP_LOGI(TAG, "CONFIG_RESET_GPIO=%d",CONFIG_RESET_GPIO);
    //i2c_master_init_ssd1306(&dev, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO, CONFIG_RESET_GPIO);
    i2c_master_init_ssd1306(&dev, 5, 4, -1);

#if CONFIG_FLIP
    dev._flip = true;
    ESP_LOGW(TAG, "Flip upside down");
#endif

#if CONFIG_SSD1306_128x64
    ESP_LOGI(TAG, "Panel is 128x64");
    ssd1306_init(&dev, 128, 64);
#endif // CONFIG_SSD1306_128x64
#if CONFIG_SSD1306_128x32
    ESP_LOGI(TAG, "Panel is 128x32");
    ssd1306_init(&dev, 128, 32);
#endif // CONFIG_SSD1306_128x32

    ssd1306_clear_screen(&dev, false);
    ssd1306_contrast(&dev, 0xff);
    ssd1306_display_text_x3(&dev, 0, "Hello", 5, false);
    vTaskDelete(NULL);
}
```

hsnprojects, 2024-09-07:
The following version works fine with the previous version of my program (originally in esp-idf 4.6).

void app_main()
{
    i2c_dev_t devRTC;
    memset(&devRTC, 0, sizeof(i2c_dev_t));

    if (ds3231_init_desc(&devRTC, I2C_NUM_0, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO) != ESP_OK)
    {
        halt("RTC failed.");
    }

    struct tm timeinfo = {
        .tm_year = 2024, .tm_mon = 7, .tm_mday = 15, .tm_hour = 17, .tm_min = 50, .tm_sec = 55};

    printf("Setting RTC to: %04d-%02d-%02d %02d:%02d:%02d\n",
           timeinfo.tm_year, timeinfo.tm_mon, timeinfo.tm_mday,
           timeinfo.tm_hour, timeinfo.tm_min, timeinfo.tm_sec);

    if (ds3231_set_time(&devRTC, &timeinfo) != ESP_OK)
    {
        halt("Error RTC Write.");
    }

    if (ds3231_get_time(&devRTC, &timeinfo) != ESP_OK)
    {
        printf("Error while reading RTC");
    }

    printf("Read back from RTC: %04d-%02d-%02d %02d:%02d:%02d\n",
           timeinfo.tm_year, timeinfo.tm_mon, timeinfo.tm_mday,
           timeinfo.tm_hour, timeinfo.tm_min, timeinfo.tm_sec);

    SSD1306_t devOled;

    devOled._address = I2C_ADDRESS; // 0x3C
    ssd1306_init(&devOled, 128, 32);

    ssd1306_clear_screen(&devOled, false);
    ssd1306_display_text_x3(&devOled, 0, "Hello", 5, false);

    while (1)
    {
        vTaskDelay(1000);
    }
}

nopnop2002, 2024-09-07:
>I am forcing it to use the legacy driver

Yes. 
Added forced legacy i2c driver mode to latest version of SSD1306.

```

$ cd $HOME

$ git clone https://github.com/nopnop2002/esp-idf-ssd1306

$ cd $HOME/esp-idf-ssd1306/components/ssd1306

$ cp CMakeLists.txt $HOME/esp-idf-ds3231/components/ssd1306/

$ cp Kconfig.projbuild $HOME/esp-idf-ds3231/components/ssd1306/

$ cd $HOME/esp-idf-ds3231

$ rm -r build

$ rm sdkconfig

$ idf.py menuconfig
```

[image]

---

Please specify the GPIO number of SSD1306 directly.
CONFIG_SDA_GPIO and CONFIG_SCL_GPIO conflict on ds3231 and ssd1306
```
void ssd1306(void *pvParameters)
{
    SSD1306_t dev;

    ESP_LOGI(TAG, "INTERFACE is i2c");
    ESP_LOGI(TAG, "CONFIG_SDA_GPIO=%d",CONFIG_SDA_GPIO);
    ESP_LOGI(TAG, "CONFIG_SCL_GPIO=%d",CONFIG_SCL_GPIO);
    ESP_LOGI(TAG, "CONFIG_RESET_GPIO=%d",CONFIG_RESET_GPIO);
    //i2c_master_init_ssd1306(&dev, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO, CONFIG_RESET_GPIO);
    i2c_master_init_ssd1306(&dev, 5, 4, -1);

#if CONFIG_FLIP
    dev._flip = true;
    ESP_LOGW(TAG, "Flip upside down");
#endif

#if CONFIG_SSD1306_128x64
    ESP_LOGI(TAG, "Panel is 128x64");
    ssd1306_init(&dev, 128, 64);
#endif // CONFIG_SSD1306_128x64
#if CONFIG_SSD1306_128x32
    ESP_LOGI(TAG, "Panel is 128x32");
    ssd1306_init(&dev, 128, 32);
#endif // CONFIG_SSD1306_128x32

    ssd1306_clear_screen(&dev, false);
    ssd1306_contrast(&dev, 0xff);
    ssd1306_display_text_x3(&dev, 0, "Hello", 5, false);
    vTaskDelete(NULL);
}
```

hsnprojects, 2024-09-07:
The reason I tried to upgrade was due to a warning I saw while building the program, which recommended upgrading the I2C driver. However, I don't see that warning now, which is strange.

nopnop2002, 2024-09-07:
Let's organize some information.
Is your ESP-IDF version 4.6?

hsnprojects, 2024-09-07:
1. I have upgraded my ESP-IDF to 5.3. I may still have 4.6 on another machine, though.

2. The version that works fine now even on 5.3 is an old clone of your repo, but the sample that I am working on now is based on your source from earlier today.

When I try to initialize I2C for the OLED with the function call 
`i2c_master_init_ssd1306(&devOled, 13, 12, -1);`

I get the following error: 
`E (345) i2c: i2c driver install error.`

However, it works fine if I remove that line.

hsnprojects, 2024-09-07:
I was hoping to use I2C in a manner similar to Arduino's `Wire.begin()`, where initializing the I2C bus would be a separate from adding devices like the RTC or OLED to the bus.

nopnop2002, 2024-09-07:
C++ is very limited in esp-idf.
In order to use the class, it is necessary to change it to main.cpp, but the driver provided by ESP-IDF does not support main.cpp.

nopnop2002, 2024-09-07:
>I get the following error:

It worked fine with the code below.
```
void app_main()
{
i2c_dev_t devRTC;
memset(&devRTC, 0, sizeof(i2c_dev_t));

if (ds3231_init_desc(&devRTC, I2C_NUM_0, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO) != ESP_OK)
{
    halt("RTC failed.");
}

struct tm timeinfo = {
    .tm_year = 2024, .tm_mon = 7, .tm_mday = 15, .tm_hour = 17, .tm_min = 50, .tm_sec = 55};

printf("Setting RTC to: %04d-%02d-%02d %02d:%02d:%02d\n",
       timeinfo.tm_year, timeinfo.tm_mon, timeinfo.tm_mday,
       timeinfo.tm_hour, timeinfo.tm_min, timeinfo.tm_sec);

if (ds3231_set_time(&devRTC, &timeinfo) != ESP_OK)
{
    halt("Error RTC Write.");
}

if (ds3231_get_time(&devRTC, &timeinfo) != ESP_OK)
{
    printf("Error while reading RTC");
}

printf("Read back from RTC: %04d-%02d-%02d %02d:%02d:%02d\n",
       timeinfo.tm_year, timeinfo.tm_mon, timeinfo.tm_mday,
       timeinfo.tm_hour, timeinfo.tm_min, timeinfo.tm_sec);

SSD1306_t devOled;

i2c_master_init_ssd1306(&devOled, 5, 4, -1);

ssd1306_init(&devOled, 128, 32);

ssd1306_clear_screen(&devOled, false);
ssd1306_display_text_x3(&devOled, 0, "Hello", 5, false);

while (1)
{
    vTaskDelay(1000);
}

}
```

```
I (332) main_task: Started on CPU0
I (342) main_task: Calling app_main()
Setting RTC to: 2024-07-15 17:50:55
Read back from RTC: 2024-07-15 17:50:55
I (342) SSD1306: Legacy i2c driver is used
I (352) SSD1306: OLED configured successfully
```

hsnprojects, 2024-09-07:
Not Object Oriented, but just c-style APIs. Initializing i2c, might create a i2c channel handle or something. Now we add devices (register devices) to one of the buses. i2c_master_init() will be called only once to initialize the bus. I can take a look at how esp-idf handles this.

hsnprojects, 2024-09-07:
> > I get the following error:
> 
> It worked fine with the code below.
> 

I will give it a try tomorrow. I may have to get latest of your source.
Thanks for working on this issue!

nopnop2002, 2024-09-07:
>Initializing I2C might create an I2C channel handle or something similar. Devices are then added (registered) to one of the buses. The i2c_master_init() function is called only once to initialize the bus. 

You're right.

However, the behavior will be as follows.
```
I2C_NUM_0 --+--DS3231 (address 0x68)
            +--SSD1307(address 0x3c)

i2c_config_t i2c_config = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = sda,
        .scl_io_num = scl,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        //.master.clk_speed = 1000000
        .master.clk_speed = 400000
};
i2c_param_config(I2C_NUM_0, &i2c_config);
return i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);

I (343) I2CDEV: DS3231 write address=0x68
I (343) I2CDEV: DS3231 write done
I (353) I2CDEV: DS3231 read address=0x68
I (353) I2CDEV: DS3231 read done
I (373) SSD1306: SSD1306 write address=0x3c
E (373) SSD1306: OLED configuration failed. code: 0xFFFFFFFF

nopnop2002, 2024-09-07:
Changed i2c_master_init to i2c_dev_init.

https://github.com/nopnop2002/esp-idf-ds3231

The i2c_master_init conflict is now gone.

hsnprojects, 2024-09-07:
Yes, that worked. I still have some issues with text rendering, but that might be related to my local setup. 

Shouldn't we be able to use the new I2C driver with ESP-IDF v5.3, rather than forcing legacy driver?

(BTW, I just have this test app all in one function for the sake of simplicity).

nopnop2002, 2024-09-08:
>Shouldn't we be able to use the new I2C driver with ESP-IDF v5.3, rather than forcing legacy driver?

Achieving this with the new i2c driver would require major changes.
```
I2C_NUM_0 --+-- DS3231  (i2c address = 0x68) 
            +-- SSD1306 (i2c address = 0x3c)
```

- Legacy driver

We only need to run this code once.
```
        i2c_config_t i2c_config = {
            .mode = I2C_MODE_MASTER,
            .sda_io_num = sda,
            .scl_io_num = scl,
            .sda_pullup_en = GPIO_PULLUP_ENABLE,
            .scl_pullup_en = GPIO_PULLUP_ENABLE,
            .master.clk_speed = I2C_MASTER_FREQ_HZ
        };
        ESP_ERROR_CHECK(i2c_param_config(I2C_NUM, &i2c_config));
        ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM, I2C_MODE_MASTER, 0, 0, 0));```
```

- New i2c driver
We need to run this code on the first device.   
We need to save dev_handle somewhere.   
```
    i2c_master_bus_config_t i2c_mst_config = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .glitch_ignore_cnt = 7,
        .i2c_port = I2C_NUM,
        .scl_io_num = scl,
        .sda_io_num = sda,
        .flags.enable_internal_pullup = true,
    };
    i2c_master_bus_handle_t bus_handle;
    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));

    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = I2C_ADDRESS1,
        .scl_speed_hz = I2C_MASTER_FREQ_HZ1,
    };
    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
```

We need to run this code on the second device.
We need to load dev_handle from somewhere.

```
    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = I2C_ADDRESS2,
        .scl_speed_hz = I2C_MASTER_FREQ_HZ2,
    };
    //i2c_master_dev_handle_t dev_handle;
    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
```

hsnprojects, 2024-09-08:
Can main() initialize i2c and store the bus handle, and pass that to each of the device initiation routines?

nopnop2002, 2024-09-08:
>then pass that to each of the device initialization routines?

You will need to add the bus handle to the initialization function argument.

hsnprojects, 2024-09-09:
Here's the revised version with minimal changes:

By this approach, you can add devices in any order, and you don't have to rely on devices to initialize the bus. This also aligns with the architecture of ESP-IDF 5.3. I think the I2C bus initialization (driver load) might have changed between versions, but I haven't looked into it.

You said:

> But We cannot add arguments to i2c_master_init as it would break compatibility.
> New functions are required to ensure compatibility

I did not quite follow this, could you explain? If arguments changed between esp-idf versions, we can include code conditionally.

nopnop2002, 2024-09-09:
```
i2c_master_init(&dev, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO, CONFIG_RESET_GPIO);
```

This function cannot change its arguments.
Because many users are using this function anyway.

hsnprojects, 2024-09-09:
That makes sense!

nopnop2002, 2024-09-09:
This function is already available.
https://github.com/nopnop2002/esp-idf-ssd1306/blob/master/components/ssd1306/ssd1306_i2c_new.c#L62

```
i2c_bus_add(&dev, bus_handle, I2C_NUM_0, CONFIG_RESET_GPIO);
```

KHAY-NOUN, 2025-01-13:
How about RTC DS1302 and BME680 sensor ?

nopnop2002, 2025-01-13:
Both work at the same time.

closed 2025-04-11 by nopnop2002