nopnop2002/esp-idf-ssd1306 #32: Long text scrolling

alexantony13, 2024-09-19:
Is there any option to display a long text, which has characters greater than the display width, by scrolling?

nopnop2002, 2024-09-19:
All you have to do is move the characters one by one and redraw them.

alexantony13, 2024-09-20:
Thanks for the prompt reply.

I was looking for a method to show a dynamic notification bar at the bottom of the screen, say 1 page.
Needed to show details like IP address, connection status, etc.

Redrawing by after shifting one character at a time will work, but the scrolling wont be smooth. I think segment wise shifting should be used. Is this correct?

Is there any builtin methods in the library for this purpose?

nopnop2002, 2024-09-20:
We understood what you want.
We plan to work on it on Monday night.

nopnop2002, 2024-09-23:
Added this project.

https://github.com/nopnop2002/esp-idf-ssd1306/tree/master/TextBoxDemo

alexantony13, 2024-09-23:
Thanks for the update.

The new feature works really well, for my purpose.

closed 2024-09-23 by alexantony13