nopnop2002/esp-idf-ssd1306 #33: Display stays black on s2 mini

mado89, 2024-09-25:
Hi there,

for the last couple of days I am trying to migrate my application from PlattformIO to ESP-IDF. However, I cannot even get the "Textdemo" application to show text on an OLED display. The display works fine, when I run my pio code but stays black (or shows garbled output) when I run esp-idf code.

Anyone here who experienced this issue in the past and could provide guidance on what I am doing wrong?

Here is the output:

```
I (30) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (31) boot: compile time Sep 20 2024 08:17:46
I (31) boot: chip revision: v1.0
I (31) boot.esp32s2: SPI Speed : 80MHz
I (31) boot.esp32s2: SPI Mode : DIO
I (31) boot.esp32s2: SPI Flash Size : 2MB
I (31) boot: Enabling RNG early entropy source...
I (32) boot: Partition Table:
I (32) boot: ## Label Usage Type ST Offset Length
I (33) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (34) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (35) boot: 2 factory factory app 00 00 00010000 00100000
I (36) boot: End of partition table
I (36) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0b5c4h ( 46532) map
I (46) esp_image: segment 1: paddr=0001b5ec vaddr=3ffc0ac0 size=023dch ( 9180) load
I (49) esp_image: segment 2: paddr=0001d9d0 vaddr=40024000 size=02648h ( 9800) load
I (51) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=1b144h (110916) map
I (74) esp_image: segment 4: paddr=0003b16c vaddr=40026648 size=0a470h ( 42096) load
I (91) boot: Loaded app from partition at offset 0x10000
I (92) boot: Disabling RNG early entropy source...
I (112) main_task: Started on CPU0
I (112) main_task: Calling app_main()
I (112) SSD1306: INTERFACE is i2c
I (112) SSD1306: CONFIG_SDA_GPIO=8
I (112) SSD1306: CONFIG_SCL_GPIO=9
I (112) SSD1306: CONFIG_RESET_GPIO=-1
I (112) SSD1306: New i2c driver is used
I (112) gpio: GPIO[8]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
I (122) gpio: GPIO[9]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
I (132) SSD1306: Panel is 128x64
I (132) SSD1306: OLED configured successfully
```

This display is connected to pins 8 / 9.

I am using external pullup resistors on SDA and SCL.

This is the display I am using:

[image]

nopnop2002, 2024-09-25:
>I am using external pullup resistors on SDA and SCL.

I've never used a pull-up resistor.

mado89, 2024-09-26:
So, it seems the display was defective. After replacing the display the code runs fine and text is displayed. Thanks for your efforts!

closed 2024-09-26 by mado89