nopnop2002/esp-idf-ssd1306 #40: Extend the GPIO range for ESP32-C3

TKWTL, 2024-12-24:
Thank for your code! It really helps a lot.

When I tried to adapt the project, I mapped the CS to GPIO20 and DC to GPIO21 in the menuconfig due to my hardware design(You know, the ESP32-C3 has a poor pin count, I had to use the UART pins.)

When I tried to save the change, the console reported an error that my setting is out of the range(0~19 for ESP32-C3), therefore the CS was still on GPIO10 and DC on GPIO3.

So I turned to the Kconfig.projbuild and removed the limitation by changing GPIO_RANGE_MAX in the file. And it works.

nopnop2002, 2024-12-24:
ESP32-C3 development boards come with and without a UART-TSB conversion chip.

[image]

On boards with a conversion chip, you can use them as normal GPIO.

These GPIOs will work without any problems when used as I2C.

On the other hand, boards without a conversion chip cannot be used as I2V.

If you use GPIO18 and GPIO19 as I2C, you will not be able to write to them again unless you manually switch the BOOT switch.

If you are making a general-purpose board, it is better not to use GPIO18 and GPIO19 on the ESP32-C3.

nopnop2002, 2024-12-24:
ESP32C3 development board comes with and without a USB-UART conversion chip.

[image]

On boards without a conversion chip, if you use GPIO18 and GPIO19 as GPIO, you will not be able to write again unless you manually switch the BOOT switch.

To avoid unnecessary trouble, GPIOs above 18 on the ESP32C3 are disabled.

TKWTL, 2024-12-24:
[image]
My schematic on EasyEDA. It's not for general purpose while its actually an embedded controller.
I think when developing ESP32C3, we always have to choose between GPIO18 19 for built-in USB-JTAG or GPIO20 21 for external USB - UART dongle chips. The built-in option would have a lower cost.

nopnop2002, 2024-12-24:
Thank you for reporting. It certainly looks like GPIO20 and GPIO21 can be used.

closed 2024-12-24 by TKWTL