nopnop2002/esp-idf-ssd1306 #46: _ssd1306_bitmaps is slow due to vTaskDelay

drewbharris, 2025-03-18:
Hi, just curious what the function of this line is: https://github.com/nopnop2002/esp-idf-ssd1306/blob/f0770f10b5d4ed037118f41a39a5f8a23d19113a/components/ssd1306/ssd1306.c#L575C3-L575C17

I took that out and `_ssd1306_bitmaps` runs about 4x faster - in the BDF demo a string draw takes 24ms instead of 115ms for me.

nopnop2002, 2025-03-18:
This is to avoid WDT alerts.

drewbharris, 2025-03-18:
I'm using an I2C SSD1306. No issues with watchdog timeout when commenting out the `vTaskDelay`. I've modified the BdfFontDemo like so ```while (1) {
  ssd1306_clear_screen(&dev, false);
  ssd1306_contrast(&dev, 0xff);

  int64_t start = esp_timer_get_time();
  show_bdf_font_text(&dev, __timR12_bitmap__, "Hello World", 0, 32);
  ESP_LOGI("", "elapsed: %llu us", esp_timer_get_time() - start);
  vTaskDelay(10);

  ssd1306_clear_screen(&dev, false);
  ssd1306_contrast(&dev, 0xff);
  vTaskDelay(10);
}```

and it works great. ~26us to draw a single line of text with `vTaskDelay` commented out. The watchdog timer is enabled - the only way I could see this being an issue is if someone is calling `_ssd1306_bitmaps` in a loop without delaying, but that should be solved in the consuming code I think - I can't see a reason to have a sleep inside a loop inside `_ssd1306_bitmaps`.

nopnop2002, 2025-03-18:
Try displaying a 128x64 image.
You'll probably see TWDT.

drewbharris, 2025-03-19:
I guess it just surprises me that this function would ever take long enough to trigger watchdog timeout (I know I can change tickrate, but in my situation I can't afford extra delay) - I will just fork and remove the timeout I think, though it might be an improvement to the API to make that optional

nopnop2002, 2025-03-19:
You can eliminate the delay in _ssd1306_bitmaps by changing the following in sdkconfig:

CONFIG_ESP_TASK_WDT_EN=n

DennisCRM, 2025-04-04:
Just seeing this comment thread.  I didn't know why vTaskDelay was in there either and just submitted a PR removing it.  However I also updated ssd1306_bitmaps to only re-draw the pages/segments that are updated by the bitmap.  This makes it **much** faster to draw any bitmap smaller than the entire screen.  I'm showing a bubble-pop animation on keypress that is crazy responsive.  Really appreciate this library as it's perfect for my project after this small change.

nopnop2002, 2025-04-06:
vTaskDelay is required for _ssd1306_bitmaps().
If you remove vTaskDelay, this error will occur.

[image]

nopnop2002, 2025-04-12:
I tested DennisCRM's PR for a week and merged this PR.

As a result, vTaskDelay(1) in _ssd1306_bitmaps is gone.

closed 2025-08-27 by nopnop2002